smith3d/Marlin #3: [BUG] Fixing center print bed with metal creality BLTouch rack

phenomeus, 2021-01-15:
fter I installed a BLTouch Kit from creality directly: https://www.creality3dofficial.com/products/creality-bl-touch?variant=31268470161481 for my Ender 3 v2

for this I had also to overwrite the probe offset with “**M851 X-45.00 Y-7.00**”

I fixed the “center print area” problem with:

print this: https://www.thingiverse.com/thing:2280529 and measure the left and right box distance to your bed. 
Y was correct and centered, only X needed adjustment in my case.

X to the left: 26mm and to the right 35mm. the difference is 9mm and this divided by 2 is 4.5mm
Then I added “**M206 X-4.5**” with prontoface and now I have centered print area.

Nicog77, 2021-01-25:
Same thing here. While using a BL Touch kit from Creality with metal bracket for the ender 3 v2, the probe is not center on the X axis.

sash231, 2021-01-27:
Hi there
I have the same problem but when I use the command M206 In my case M206 X-8.3 Y-1.0. the BLTouch is in the middle but the injection line is too far to the right and my X stepper motor makes strange noises when nothing is Moving.

can someone help me there please? :) thanks

JohnClementine, 2021-01-28:
This helped me get my z home perfectly center so thank you very much, but regardless of what parameters I use with M206 and M851 the actual bed levelling itself is totally off (probing way below and to the left of where it should be to be symmetrical).

 I've even tried setting ridiculous offset values like x -50 y-50, the homing feature reflects the offsets I set but the actual levelling command probes the same points regardless of how I set up my offsets. 

Does anyone know how to resolve this?

phenomeus, 2021-01-29:
that's something only @smith3d can answer. thanks in advance

smith3d, 2021-01-30:
> This helped me get my z home perfectly center so thank you very much, but regardless of what parameters I use with M206 and M851 the actual bed levelling itself is totally off (probing way below and [...]

Seems like it doesn't save. Have you tried M500?