smith3d/Marlin #4: [BUG] center probing area

phenomeus, 2021-01-15:
I used 3x3 slow and now I'm testing 5x5 fast and I see that the probe area is not centered during leveling. 
is this normal or also something to modify with an offset?

JohnClementine, 2021-01-28:
I'm also having this same issue. I was able to set my bed offsets and nozzle to probe offsets to get it to home perfectly in the center, but the actual probing command is still way off center

Were you able to find a resolution?

phenomeus, 2021-01-29:
nope, but the prints are still ok

weinrank, 2021-02-02:
Im facing the same problem.
Offsets are correct, EEPROM reset didn't help.

I've also tried to recompile the firmware, no improvement.
While the "home" Position at 0,0 seems to be okay the end positions of the 5x5 grid are totally off by multiple cm.
This seems to only affect the X-axis. The Y-axis offset looks fine.

samsp-msft, 2021-03-23:
I have had this issue with I think all versions of the smith3d firmware. Its as if the probing logic is not accounting for the probe offset values when calculating the coordinates to perform measurements.