smith3d/Marlin #8: [BUG] M84 is not respecting flags

eblieb, 2021-01-26:
My Cura End G-Code has 
M84 X Y E ;Disable all steppers but Z

But this isn't being respected, the Z stepper is still disabling after the print is done. According to Merlin documentation M84 and M18 should be the same exact thing and should respect the flags.

DanielLaberge, 2021-03-08:
Confirmed. Tested both v2.0.x.16 and latest code.

I've had a look at M17_M18_M84.cpp and the code looks good. It's also the same as upstream marlin so I don't know where this behaviour is coming from.

To reproduce: 
1. M17 ; Enable Steppers
2. M18 X; Disable X-axis stepper

Result: All steppers are disabled, not just X as expected.

eblieb, 2021-03-08:
Turns out the ender 3 only uses 1 source for all the stepper motor power so
it is impossible to just turn off select ones

DanielLaberge, 2021-03-08:
This explains that. Thank you.

This bug should be closed as it cannot be resolved in software.