smith3d/Marlin #9: [BUG] Offsets are set perfectly but G29 bed level probing is very off center

JohnClementine, 2021-01-28:
I'm able to get my printer to home perfectly in the center of my bed by using the M206 and M851 commands to set my probe and bed offsets, however regardless of what I set my offsets to the actual bed level test probing invoked by G29 is waaaaaay off center/not symmetrical and doesn't seem to be influenced at all by my offsets. The probe goes almost off the bed at the bottom, and probes way more to the left than to the right.

Am I missing something? I thought that changing my offsets is also supposed to change the probing behavior.. My Z Homing is clearly influenced by the offsets, but I can't get it to bed level test any differently even if I use crazy offsets like -50 -50.

I greatly appreciate any advice at all. I've tried looking it up extensively but the only advice I've gotten is making sure my offsets are correct, which I've measured by hand as well as confirmed by the fact that when I home it probes perfectly in the center of the bed.

Heli0saurus, 2021-02-17:
Also having the same issue. Just installed a brand new E3D v6 and a Hero Me duct, triple checked the probe to nozzle offsets in the console and measured just as many times, the provided offsets and my measurements both check out. G28 homes dead center, G29 is in the front left corner 2mm from probing off the bed. Looked through config and config_adv.h and found nothing pertaining to probe and hotend positioning during homing/ABL, all the offsets were correctly set.

JohnClementine, 2021-02-17:
> Also having the same issue. Just installed a brand new E3D v6 and a Hero Me duct, triple checked the probe to nozzle offsets in the console and measured just as many times, the provided offsets in the Hero Me manual and my measurements both check out. G28 homes dead center, G29 is in the front left corner 2mm from probing off the bed. Looked through config and config_adv.h and found nothing pertaining to probe and hotend positioning during homing/ABL, all the offsets were correctly set.

Hey friend,

I got this resolved but I didn't test it with Smith3D Marlin, I have it working on the current most recent Marlin Bugfix branch.
I found out that setting the bed offsets using GCode commands does not change the probing behavior, I had to set the following in my configuration.h: 

`#define X_MIN_POS -11 
#define Y_MIN_POS -11 //my x_min_pos and y_min_pos were identical to eachother at -11, yours may differ
#define X_MAX_POS (X_BED_SIZE + 15) //I don't remember how I figured out it was +15, I think I had to load firmware with a high limit and add +1 past 220 (bed size) until I saw the hotend couldn't go any further and started grinding/hitting the side
#define Y_MAX_POS (Y_BED_SIZE + 5) //same as above`

Once I did that, my probing was perfectly symmetrical across the bed.

Here is the guide I used, good luck!
https://manuelmclure.github.io/ConfiguringLeveling.html

closed 2021-02-17 by JohnClementine