smith3d/Marlin #13: ender 3 v2 extender

geedog66, 2021-02-04:
I have your firmware downloaded on my ender 3 v2 printer and i recently purchased the ender extender kit which makes my printer 400 x 400 x 500.. I still want to keep the features in your current firmware just be able to use it on my upgraded printer.. any help will be appreciated thanks..

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use one of the support links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

Before filing an issue be sure to test the "bugfix" branches to see whether the issue has been resolved.

-->

### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

jpbarcellos, 2021-02-12:
[image]

dcwalmsley, 2021-02-24:
@geedog66 

Do you have a BLTouch installed?  I am building off Smith3D's master a 400x400x250 version.  I can compile one for you with 400x400x500 if that will work with you.

jpbarcellos, 2021-02-24:
yes, i do have bl touch.  I already did the compilation and everything is working normal.  the only thing I couldn't get it to work on was the filament sensor.  Can you help me with this? My Ender is v2 thanks