smith3d/Marlin #14: Your latest branch bugfix2.0.x-Smith3D for Ender3 V2 (06 february 2021) completely bugged !

tititopher68, 2021-02-06:
### Bug Description

If we compile with this content, we get many errors :

_In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2425:6: error: #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
     #error "**LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2**."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2425:6: error: #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
     #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\SPI.cpp:43:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2425:6: error: #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
     #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2425:6: error: #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
     #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2425:6: error: #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
     #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_flash.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2425:6: error: #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
     #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:55:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/../core/serial.h:67:34: error: 'MYSERIAL' was not declared in this scope
     typedef MultiSerial<decltype(MYSERIAL), decltype(SERIAL_CATCHALL), 0> SerialOutputT;
                                  ^~~~~~~~
Marlin\src\HAL\STM32F1\../../inc/../core/serial.h:67:34: note: suggested alternative: 'MYSERIAL1'
     typedef MultiSerial<decltype(MYSERIAL), decltype(SERIAL_CATCHALL), 0> SerialOutputT;
                                  ^~~~~~~~
                                  MYSERIAL1
Marlin\src\HAL\STM32F1\../../inc/../core/serial.h:67:34: error: 'MYSERIAL' was not declared in this scope
Marlin\src\HAL\STM32F1\../../inc/../core/serial.h:67:34: note: suggested alternative: 'MYSERIAL1'
     typedef MultiSerial<decltype(MYSERIAL), decltype(SERIAL_CATCHALL), 0> SerialOutputT;
                                  ^~~~~~~~
                                  MYSERIAL1
Marlin\src\HAL\STM32F1\../../inc/../core/serial.h:67:73: error: template argument 1 is invalid
     typedef MultiSerial<decltype(MYSERIAL), decltype(SERIAL_CATCHALL), 0> SerialOutputT;
                                                                         ^
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp:21:
Marlin\src\HAL\STM32F1\dogm\../../../inc/SanityCheck.h:2425:6: error: #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
     #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_bl24cxx.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2425:6: error: #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
     #error "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:55:0,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/../core/serial.h:67:34: error: 'MYSERIAL' was not declared in this scope
     typedef MultiSerial<decltype(MYSERIAL), decltype(SERIAL_CATCHALL), 0> SerialOutputT;
                                  ^~~~~~~~
Marlin\src\HAL\STM32F1\../../inc/../core/serial.h:67:34: note: suggested alternative: 'MYSERIAL1'
     typedef MultiSerial<decltype(MYSERIAL), decltype(SERIAL_CATCHALL), 0> SerialOutputT;
                                  ^~~~~~~~
                                  MYSERIAL1
[... 128 lines cut ...]

**Question :** 

What is the use of **#define SERIAL_PORT_2 3** in your Configuration.h file  ?

_

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

jpbarcellos, 2021-02-12:
Hey man,

Did you try to use version 14 of the Smith3d code?  Checks if your Visual Studio Code has PIO updated to T SMT32.  mine was with error in upgrade 11 I uninstalled and I'm using 10. I managed to compile the code without error.

smith3d, 2021-02-12:
just updated. i have removed the serial port

closed 2021-02-15 by smith3d