smith3d/Marlin #16: [BUG] Something is very off with BLtouch in Alpha

jasonmacdonald, 2021-02-13:
### Bug Description

There seems to be a strange issue with the newest codebase. When I compare bed leveling grid numbers between stable and alpha, the alpha ones are way off, despite changing nothing but the firmware.

Stable 4 corners (My Ender 3 v2 has a very warped bed despite the glass)

FrontR 0.065
FrontL 0.342
BackR 0.392
BackL 0.353

After firmware upgrade

FrontR 0.182
FrontL 0.3295
BackR 0.6145
BackL 0.477

I'm not sure what the pattern is here but it causes the back corners to be way off the bed when printing using the new firmware.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Using current stable firmware, do auto bed leveling and record grid numbers
2. Reload printer with the new Alpha version compiled from source and run bed leveling again
3. Record gird numbers.
4. Go back to the stable version and repeat.
5. Numbers for Alpha  are way off when compared to both stable sets

**Expected behavior:** [What you expect to happen]
Hey should be fairly similar

**Actual behavior:** [What actually happens]
The back corners are way off causing prints to be too high off the bed.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

wob8, 2021-02-14:
Same observation, returned to previous stable version because of this.

Nicog77, 2021-02-14:
Same behiavior here. My bed is flat and abl seems to rise the head too much from the bed when printing the first layer. I use the M420 S1 Z2 as starting code in cura.
[image]

jasonmacdonald, 2021-02-14:
Just a note, I think it's Marlin's fault. I switched to compiling just directly from Marlin and I still have the same issues. Others are starting to claim this same https://github.com/MarlinFirmware/Marlin/issues/20922 

I wish I could figure out how to go back to the version I had before, but I have to compile it myself to compensate for my Microswiss direct drive.

smith3d, 2021-02-15:
Can you try to print without Octoprint? I'm using .17 version and it's working fine here.

jasonmacdonald, 2021-02-16:
Does the Z offset tool take the mesh into account? Because I set the z-offset using the tool, it's perfect. Then I go out, disable steppers and move to back right, and check, perfect. Start a print, it goes to the back corner and it's 1-2mm too high. It don't get it. It's been days of this. Pull my machine apart twice trying to figure this out.

Would it be possible for you to compile x.17 version with the X_min and Y_min set to -5 and -10 for MS DD so I can try testing it? Perhaps it's something I'm doing in the compile, but every time I level the bed and run the offset tool everything seems like it should be good. But when I start a print the back corners are suddenly offset by like 0.1 to 0.2 from where they should be.

smith3d, 2021-02-16:
Z offset tool is to rehome z + offset. it doesn't take mesh into account.

jasonmacdonald, 2021-02-16:
> Z offset tool is to rehome z + offset. it doesn't take mesh into account.

ok, thanks. I'm running out of things to try and figure this out. Could my BLTouch just be broken? I mean it seems fine when I probe and the mesh looks like my slightly warped bed, but when it prints something is very off.

jasonmacdonald, 2021-02-16:
Ok, more strangeness. I upgraded again to the latest and ran `M48` multiple times in the back right corner, never changing any.

First-round it reported:
```
Mean: 0.409917 Min: 0.356 Max: 0.463 Range: 0.107
Standard Deviation: 0.034312
```

By the 6th round it reported:
```
Mean: 1.281750 Min: 1.231 Max: 1.335 Range: 0.104
Standard Deviation: 0.033322
```

A change of almost a full millimeter, without touching a single thing. Yet, the standard deviation between rounds stayed more or less the same.

Here's the full log.
```
Recv: Mean: 0.409917 Min: 0.356 Max: 0.463 Range: 0.107
Recv: Standard Deviation: 0.034312
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 0.546750 Min: 0.493 Max: 0.602 Range: 0.109
Recv: Standard Deviation: 0.034642
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
[...]
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 0.682833 Min: 0.628 Max: 0.738 Range: 0.110
Recv: Standard Deviation: 0.034267
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
[...]
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 0.820167 Min: 0.219 Max: 0.934 Range: 0.715
Recv: Standard Deviation: 0.202422
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
[...]
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 0.954500 Min: 0.353 Max: 1.067 Range: 0.714
Recv: Standard Deviation: 0.202299
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
[...]
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 1.281750 Min: 1.231 Max: 1.335 Range: 0.104
Recv: Standard Deviation: 0.033322
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
[...]
```

smith3d, 2021-02-17:
do you have another machine with bltouch lying around it would be great if we can cross check if it's causes by bltouch pin.

jasonmacdonald, 2021-02-17:
I wish I did, but I do not. :(

smith3d, 2021-02-18:
version 17
test 1
Mean: -0.005300 Min: -0.008 Max: -0.002 Range: 0.006
Standard Deviation: 0.001965

test 2
Mean: -0.005450 Min: -0.007 Max: -0.003 Range: 0.003
Standard Deviation: 0.001254

test 3
Mean: -0.005750 Min: -0.008 Max: -0.003 Range: 0.005
Standard Deviation: 0.001662

version 16

test 1
Mean: -0.008550 Min: -0.011 Max: -0.005 Range: 0.006
Standard Deviation: 0.001781

test 2
Mean: -0.010600 Min: -0.012 Max: -0.008 Range: 0.004
Standard Deviation: 0.001136

test 3
Mean: -0.010650 Min: -0.014 Max: -0.007 Range: 0.007
Standard Deviation: 0.001790

here's my result for both versions.

maybe you should try the recharge your bltouch pin

Nicog77, 2021-02-18:
I remenber there is an accuracy test command for probing : M48. May this help. 
Unfortunately I can't test. I roll back to version 16 because I have a lot of print to do.

jasonmacdonald, 2021-02-18:
> maybe you should try the recharge your bltouch pin

I'm not sure what this means. How do you recharge a pin? Do you mean to change the pin to the z-stop?

jasonmacdonald, 2021-02-21:
So, it turns out this was my fault. My Z lead screw was ever so slightly loose, just enough so that with the torque from the motor it would slip, but would happen when turning it by hand. Once I tightened the screws up, as well as the ferrite screw in the BLTouch, I seem to be getting much more accurate probes.

I'll close this for now since I'm no longer having the issue.

closed 2021-02-21 by jasonmacdonald