smith3d/Marlin #18: [FR] Configuration files for the pre-compiled firmware you publish

DanielLaberge, 2021-02-18:
Are the configuration files and build settings you use to create the pre-compiled firmware files you publish _(For example, [this release for the Ender 3 v2](https://github.com/smith3d/Marlin/releases/tag/2.0.x.16))_ available anywhere? If not, can they be made available?

I appreciate that your releases offer different preset mesh grid configurations, but I would like to use Smith3D-E3V2-2.0.x.16-5x5-Fast-100121.bin as a starting point to tune some features and configuration values. 

Thanks for your consideration and time.

ely-az, 2021-02-19:
I second this! Better yet provide the config files for all the available variants. I have an Ender 3v2 with the 32bit 4.2.2 board, a BL Touch and a BTT Smart Sensor. I can't find any info about what the different firmware versions you are providing do?

smith3d, 2021-02-21:
I've attached them inside my changelog.txt

latest: configuration: https://github.com/smith3d/Marlin/blob/bugfix-2.0.x-Smith3D/Marlin/Configuration.h

following config changes based on the firmware.

#define FILAMENT_RUNOUT_SENSOR
#define NOZZLE_TO_PROBE_OFFSET
#define BLTOUCH_HS_MODE
#define GRID_MAX_POINTS_X
#define X_BED_SIZE

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

//For Standard Filament Runout Sensor
#define FIL_RUNOUT_STATE     HIGH 

//For BTT Filament Smart Sensor
#define FIL_RUNOUT_STATE     LOW 
#define FILAMENT_RUNOUT_DISTANCE_MM 7
#define FILAMENT_MOTION_SENSOR
//For RaspBerry BTT
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT

Change: #define MOTHERBOARD BOARD_CREALITY_V4
To: #define MOTHERBOARD BOARD_CREALITY_V427

//All Metal Max Temp #define HEATER_0_MAXTEMP 300

closed 2021-03-19 by smith3d