smith3d/Marlin #25: [FR] source with configs ?

spacecabbie, 2021-03-11:
Hi and hello,

I was wondering if the source could include the changed 
Configuration_adv.h
Configuration.h
or other files
For Ender 3 v2 with 4.2.7. board.

Love to compile it my self to make changes and see how it was done so I understand firmware even better.
thanks !

smith3d, 2021-03-18:
https://github.com/smith3d/Marlin/blob/bugfix-2.0.x-Smith3D/Marlin/Changelog.txt

the changes can be found inside the changelog, bottom the definition part

closed 2021-03-18 by smith3d

spacecabbie, 2021-03-18:
Thanks for your reply that shows changes added unless i am missing something.
what I mean is this:
this file: https://github.com/smith3d/Marlin/releases/download/2.0.x.17/Smith3D-E3V2-2.0.x.17-427-Fast-5x5-180321.bin

Is build with changes made in the configs:
https://github.com/smith3d/Marlin/blob/bugfix-2.0.x-Smith3D/Marlin/Configuration.h
https://github.com/smith3d/Marlin/blob/bugfix-2.0.x-Smith3D/Marlin/Configuration_adv.h
The ones I can find in  the source and on github show only the configs for 4.2.2 board.
I would realy like to get the 2 above files to compile the mention firmware bin file.

smith3d, 2021-03-19:
Change: #define MOTHERBOARD BOARD_CREALITY_V4
To: #define MOTHERBOARD BOARD_CREALITY_V427

that's the only config we changed. you can search for #define MOTHERBOARD BOARD_CREALITY_V4 & replace