smith3d/Marlin #26: Changing Z-Offset is not moving the Z-Axis

tfaz099, 2021-03-15:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

smith3d, 2021-03-16:
usually it meant incorrect wiring. check ur bltouch wiring sequence, and reseat your bltouch wiring.

tfaz099, 2021-03-27:
I think the resulting code / option change is : 
//#define AUTO_BED_LEVELING_BILINEAR  //comment out
#define AUTO_BED_LEVELING_UBL //uncomment supposedly this allows for active Z-Axis move live without having to re-home.  I will test soon.

tfaz099, 2021-03-28:
The "always use babysteps" was commented out, works fine now.

closed 2021-03-28 by tfaz099