smith3d/Marlin #27: [BUG] (Calibration Extruder)

javierzg, 2021-03-19:
Calibration Extruder
Greetings I hope to be useful with my problem and installed the new Firmware V2.0.x.17 Smith3D and had problems when making compensation of the Extruder, I explain.
_Firstly when entering x data in control-Motion-Steps / mm-Steps / mm E leaving an x value to later go to Prepare-Move-Extruder, when starting to move the knob and before being able to leave it 100.0 the extruder alone and without giving it no parameter begins to retract without being able to change a parameter or exit Move Axis, when retracting a quantity of this filament, it only begins to extrude the marked parameter that remains pasted on the screen. ???.

### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tfaz099, 2021-03-28:
@javierzg I read your issue but having a hard time myself trying to understand what you are trying to accomplish.  Can you please give some brief detail as to what you are trying to do? 
thx

javierzg, 2021-03-29:
Greetings
Smit3d.
Thank you for your new Firmware and installed from version 14 onwards and
very happy with your updates.
I rewrite my problem, hope you can review it.
I think my previous comment was not understood or I did not know how to
explain, Maybe the translation is not my strong point, my apologies.
I have an Ender 3 V2 with 4.2.2 board and dual gear extruder. Without
filament sensor, I installed the new firmware 4.2.2
(Smith3D-E3V2-2.0.x.17-Fast-5 × 5-180321.bin) by default.
I have problems when making the compensation of the extruder, previously it
had compensated in Steps / mmE 140.0
With the new firmware, when you want to enter new data in Steps / mm E
leaving the same value or another and then move to Move-Extruder, when you
start moving the knob the extruder alone and without giving any parameter,
the extruder filament begins to retract without To be able to change a
parameter or exit Move Axis, if you stop moving the knob by marking 50 mm
or 100.0 mm the screen sticks but continues to remove the filament, it only
begins to extrude the marked parameter that remains stuck on the screen.
once it has finished retracting a quantity of filament ???. Due to this
problem, I had to go back to the previous version installed, the older
firmware V2.0.X.16.
I would appreciate your comment.
Thanks.