smith3d/Marlin #28: [BIG] Smith3D-E3V2-2.0.x.17-Fast-5x5-180321.bin USB connection from octoprint not working

alexeyatbluescape, 2021-03-25:
### Description

Latest build for board 4.2.2:  Smith3D-E3V2-2.0.x.17-Fast-5x5-180321.bin breaks USB connection with octoprint.

### Steps to Reproduce

1. Flask Smith3D-E3V2-2.0.x.17-Fast-5x5-180321.bin
2. Test commands via usb

**Expected behavior:** [What you expect to happen]

Commands work

**Actual behavior:** [What actually happens]

They don't.

#### Additional Information

Error in octoprint terminal:

```
Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
```

reverted to a previous release: Smith3D-E3V2-2.0.x.14-3x3.bin

everything works.

tfaz099, 2021-03-28:
Which board are you using?

alexeyatbluescape, 2021-03-28:
4.2.2 seems to be the right one

tfaz099, 2021-03-28:
[firmware-20210324-210154.bin.zip](https://github.com/smith3d/Marlin/files/6216698/firmware-20210324-210154.bin.zip)

Try this one.  It is the latest compiled for a BLTouch (If you do not have a BLTouch do not use this)

I have been running it for 2 days and it has been flawless with respect to basic operation.  Note: this is a user compiled version of the Smith3D GitHub repository i am not affiliated with Marlin or Smith3D.  It works for me.

alexeyatbluescape, 2021-03-28:
Thanks @tfaz099 I will tomorrow, once current print is done. I do have BLTouch, this should work. How are the configs across all version here are persisted? I with I could get one that I like and play with couple settings. Does yours have m600 command as well?

tfaz099, 2021-03-28:
Note, this is probably not the best place to have a discussion, we can move it to the "discussions" but to finish this off...
My start script (i am using Octoprint also and just recently started running off an SSD) any more discussion please start a thread in discussions and tag me)

G28 ; home all axes
M92 E140 ;e-steps <- your e-steps are likely different
M420 S ;Enable ABL using saved Mesh
G1 X15 Y20 Z0.2 F3000  ; get ready to prime
G92 E0 ; reset extrusion distance
G1 X160 E15 F600 ; prime nozzle
G1 X180 F5000 ; quick wipe

philipphoh, 2021-03-28:
I encountered the same problem. Octoprint can't connect to the printer with "Error: No more candidates to test, and no working port/baudrate combination detected." 
When connecting the printer to my PC it shows an unknown USB Serial connection in the device manager, with no compatible drivers being available. 

I flashed the firmware from the marlin firmware service and it worked "fine", at least in regard of connecting to and controlling the printer. Other Issues with that firmware stemmed from the "nightly"-nature of those builds, not related to the problems I have with the Smith3D release. 
The user-compiled firmware provided in the post above doesn't seem to fix the problem either.

philipphoh, 2021-03-28:
I have now solved the problem by using the `Smith3D-E3V2-2.0.x.17-Fast-5x5-230x230-180321.bin` and changing the detection baud rate in octoprint to 250000. 
Evering seems to be working, I'm running my first testprint right now.

tfaz099, 2021-03-28:
@philipphoh @alexeyatbluescape if you both have solved your issue, please close this issue so the actual developers have less on their plate.  thank you.

alexeyatbluescape, 2021-03-29:
@tfaz099 Can confirm, when changing the detection baud rate in octoprint to 250000, it works. Thanks @philipphoh for the hint! Not sure why auto was not detecting that rate before, now seems to be fine. Using `Smith3D-E3V2-2.0.x.17-Fast-5x5-180321.bin`

closed 2021-03-29 by alexeyatbluescape