smith3d/Marlin #33: filament runout sensor

menthol1960, 2021-03-31:
Hello

I have an issue with the filament run out.
I am using E3V2-2.0.x.17-Fast-5x5-180321 version 
cable checked metered etc.

The issue is that it cleans nozzle 
Then goes to print and the run out sensor triggers even though filament still present 
any advice would be appreciated

OhadHenn, 2021-03-31:
I have this issue two, same printer

defetzer, 2021-04-02:
same, also with the inverted version.No print possible.

arxcruz, 2021-04-02:
Same problem, I even tried to change the FIL_RUNOUT_STATE to LOW but same problem

Sersch97, 2021-04-05:
same here, tried the normal and the inverted version but it still always retracts and waits

ulivero, 2021-04-05:
Hi , I have the same problem too, i have an ender 3 v2 , board 4.2.2 with bltouch and folament runout sensor 
Withe the version of firmware slow, fast and inverted they all do the same , start print pause the print unload the filament , its not working , theres any solution to this ? i think the have to compile another firmware with this fix .
any have a solution to this?
thanks in advance

Thierry-56, 2021-04-05:
I am using Smith3D-E3V2-2.0.x.17-427-Fast-5x5-180321  version and I have the same problem:
 start print,  pause the print and unload the filament 
Thanks in advance

galluccib, 2021-04-05:
Same here.... start print, pause the print and unload the filament

galluccib, 2021-04-05:
Same on 4.2.2 - Smith3D-E3V2-2.0.x.17-427-Fast-5x5-180321

ulivero, 2021-04-07:
Hi again , anybody has a fix for this of filament runout pause the print and unload the filament ?
Thank you so much

rembhu, 2021-04-08:
... and another one with this problem who waits for solution.

N3tYaroz3, 2021-04-11:
I'm here for it too :)

Zaaap, 2021-04-12:
Same here for Smith3D-E3V2-2.0.x.17-Fast-5x5-180321.bin

vcrproszek, 2021-05-24:
I had the same problem. Simth recommend to use Jyers firmware now, so I did it, and now sensor is working properly https://github.com/Jyers